kaidyth/nrf52_bootloader #2: Document how to flash Arduino Sketch

charlesportwoodii, 2019-06-25:


charlesportwoodii, 2019-06-26:
@dsteinman Yes, you can flash an Arduino sketch:

1. Export Compiled Binary on your sketch
2. Generate a signed package:
```
 nrfutil pkg generate --hw-version 52 --application-version 1 --application /path/to/compiled/program.hex --sd-req 0xB6 --key-file /path/to/kaidyth_nrf52_bootloader/default_private.pem app_dfu_package.zip
```
3. Flash over USB
```
nrfutil dfu usb-serial --package app_dfu_package.zip -p /dev/ttyACM0
```

I tested this with a simple program that just dumped text to a serial port in a loop and monitored it via [tio](https://tio.github.io/) on Linux and verified the sketch was running.

closed 2019-06-26 by charlesportwoodii

uttaravadina, 2021-08-09:
This is confusing a bit. Can  you tell me if this is possible?

- Use Nordic's official softdevice hex file
- Create secure bootloader on Nordic's sdk
- Create a sketch and use [BLuefruit](https://github.com/adafruit/Adafruit_nRF52_Arduino) library to build an Application

uttaravadina, 2021-08-09:
@charlesportwoodii I would really appreciate your answer here.

charlesportwoodii, 2021-08-09:
Answers to each of these questions are documented in the README. Please review the README file, and if you have additional issue create a new issue rather than commenting on unrelated and old issues.