kaidyth/nrf52_bootloader #12: Does this bootloader support nRF52832?

rabih9780, 2019-07-10:


charlesportwoodii, 2019-07-10:
No, it does not.

Adding support for it wouldn't be too difficult, you basically just need to add the linker files and update the makefile targets for that specific board, and there wouldn't be an double-tap-reset support.

Pull requests are welcome. I don't have any nRF52832 boards at my disposal to test with.

closed 2019-07-10 by charlesportwoodii

rabih9780, 2019-07-10:
Thank you for the quick response. 

Can you please quickly comment on the main difference between your bootloader vs adafruit nrf52 bootloader (https://github.com/adafruit/Adafruit_nRF52_Bootloader)?

charlesportwoodii, 2019-07-10:
This is documented in the readme but for clarity, it has the same feature/board support as the Adafruit one with the following differences:

- Supports Nordic Secure BLE DFU OTA instead of the Legacy one
- Uses the standard nrfutil tools instead of the custom Adafruit ones.
- Presently doesn't have support for anything other than nRF52840 chipsets
- Does not support UF2 yet

In short, I needed the Secure BLE DFU OTA functionality for an app I have and the Adafruit one didn't provide it so I made this.

rabih9780, 2019-07-10:
If I end up using your bootloader and added support for nrf52832, do you know if this bootloader will work with Adafruit_nRF52_Arduino (https://github.com/adafruit/Adafruit_nRF52_Arduino)?

charlesportwoodii, 2019-07-10:
Unfortunately it doesn't due to an upstream issue.  The normal arduino stuff works, the Adafruit Bluefruit libraries however do not. See https://github.com/charlesportwoodii/kaidyth_nrf52_bootloader/issues/3#issuecomment-507072763 and adafruit/Adafruit_nRF52_Arduino#200.

It's a problem with how the Bluefruit libraries interact with the Nordic Secure Bootloader which this is based off of. Adafruit would need to update the Bluefruit libraries to make them compatible.