kaidyth/nrf52_bootloader #14: Initial attempt at building with Docker fails

SaltedBlowfish, 2020-06-16:
Stumbled across this repo while working on my own nRF52840 bootloader for a project. I was excited to see that someone had already worked on a solution but when I tried to build the Docker image, it failed. Details:

#### End of output:

```bash
make: Entering directory '/root/nrf_sdk/16.0.0/external/micro-ecc/nrf51_armgcc/armgcc'
make: /root/gcc-arm-none-eabi-9-2019-q4-update/bin/arm-none-eabi-gcc: Command not found
Cannot find: '/root/gcc-arm-none-eabi-9-2019-q4-update/bin/arm-none-eabi-gcc'.
Please set values in: "/root/nrf_sdk/16.0.0/components/toolchain/gcc/Makefile.posix"
according to the actual configuration of your system.
../../../../components/toolchain/gcc/Makefile.common:129: *** Cannot continue.  Stop.
make: Leaving directory '/root/nrf_sdk/16.0.0/external/micro-ecc/nrf51_armgcc/armgcc'
The command '/bin/sh -c chmod +x /tmp/before_install.sh &&     ./tmp/before_install.sh &&     pip install --user nrfutil' returned a non-zero code: 2
```

Any advice is appreciated.

charlesportwoodii, 2020-06-16:
Sorry about that, I haven't updated the Dockerfile in a while but it should be easy to fix once I have time to sit down and take a look.

I recommend complaining locally instead until I can fix the Dockerfile - just follow the normal SDK setup steps Nordic publishes for your platform.

I do have pre-complied binaries at https://github.com/kaidyth/nrf52_bootloader/releases you can download. If you're just looking for a build use one of those - they're the ones I use for all of my boards that I test on.

I'll follow up once I've had a chance to update the Dockerfile to the latest SDK in use.

charlesportwoodii, 2020-06-17:
Looks like ARM changed the download location. Latest commit fixes the issue.

commit 9dcc7c52 (2020-06-17): Fixes #14

ARM GCC EABI download wasn't working after update
closed 2020-06-17 by charlesportwoodii