kaidyth/nrf52_bootloader #15: Application flashes successfully, does not load

SaltedBlowfish, 2020-06-29:
### Summary

This issue (https://github.com/kaidyth/nrf52_bootloader/issues/3) is very similar to what I'm seeing, but I have some questions about the conclusions reached in that issue. I'm in no way asking for troubleshooting steps to solve my unique case, but I'd like to find out what some of the findings of that issue mean to those of us implementing this bootloader on a Nano 33 BLE or other board based on the u-blox Nina B306.

First off:
1. I'm using a custom board built on the u-blox Nina B306 (same exact SoC as the Arduino Nano 33 BLE)
1. I can flash the nRF52 Adafruit bootloader **and** applications built with the Adafruit BSP for development without any issues. All functionality from Adafruit works, including legacy DFU, USB uploading from Arduino, PlatformIO, USB Serial, etc.
1. I can flash the Kaidyth bootloader and flash signed applications via USB and BLE without issues, **but the application flashed in this manner does not load**.

### Reproducing

#### Bootloader

Start by building and flashing the `Feather Express Bootloader` with default settings.

#### Application Code

```
void setup() { Serial.begin(115200); }
void loop()
{
  Serial.println("loop");
  delay(1000);
}
```

#### Build Application Package and Flash

1. Arduino IDE > Sketch > Export compiled Binary > `nrfutil pkg generate [...]` > `nrfutil dfu usb-serial [...]`
1. At this point, the board resets but the application never loads and it's not visible in the USB devices anymore.

### Questions

1. Is the conclusion that applications generated (using the Feather nRF52840 Express BSP) are incompatible with this bootloader or only if the `<bluefruit.h>` library is explicitly included? (My results seem to indicate that regardless of including the bluefruit library, it fails, but it might not be the Adafruit libraries in my case)
1. If item 1 is the case, is there another BSP suggested for use with this bootloader to generate the application packages? (I've also tried the Arduino Nano 33 BLE BSP and it failed in the same manner with the `generic` bootloader variant)

charlesportwoodii, 2020-06-29:
Hi,

If I recall, it was just the bluefruit library that was causing a hang - I was able to get a variety of different Arduino sketches to work including one that did an RBG ring of the embedded LED on the Feather, but looking back on my notes I do remember I had issues with serial outs.

At it's core, this bootloader really is just the default Nordic secure bootloader - it just has all the features enabled for DFU and a double reset functionality built in.

I can take another look though since it's been a while.

A couple of questions:

1. Can you share your exact try command you used for flashing and building? I upgraded the SD version so if you copied the command from #3 it _should_ fail cause of an SD mismatch. The SD should be 0xCA when signing.
2. Does the Nano have a LED you can flash instead of writing to serial? GPIO was what I was testing and it's a bit easier to validate that way since it doesn't require an external device or connectivity to see if the sketch is doing anything.
3. The generic bootloader runs in debug mode - do you have a JLink or similar tool you can connect to read the debug serial output and share here? The debug logs can indicate if the bootloader is able to pass off to the application or not. If not... You can recompile the generic bootloader to spit out UART logs to any pins of your choosing and capture them from a generic UART reader. Just make sure you compile in DEBUG mode.
4. Before you flash the application can you confirm that you see KAIDYTH_DFU in nrfconnect? 
5. Does the double tap to reset work or the Nano _after_ your application is flashed? If there's not a button, you should be able to short and of the reset pins to GND twice and see KAIDYTH_DFU appear in nrfconnect. If this doesn't work then that means there's a signing problem with the app most likely.

I can order the Nano 33 BLE and have it rush shipped to debug it correctly. I need a new chip to play with while my custom board I'm working on needs revisions. =)

Arduino isn't really something I've dived too deep in - I went straight to C instead, but I can certainly investigate this with you.

SaltedBlowfish, 2020-06-29:
@charlesportwoodii Thank you for the prompt reply. Here are the answers to your questions:

1. Commands:

```
nrfutil pkg generate --hw-version 52 --application-version 1 --application ~/Documents/Arduino/test001/test001.ino.feather_nrf52840_express.hex --sd-req 0xCA --key-file /path/to/private.pem app_dfu_package.zip
```

```
nrfutil dfu usb-serial --package app_dfu_package.zip -p /dev/cu.usbmodemE23B6AAE27A81
```

2. Yes, I attempted to flash the LED but to no avail.
3. I do have a JLink. I'll report back later.
4. Absolutely. I'm able to perform secure BLE DFU. The board is also listed in the USB devices on my computer while in the DFU mode.

[image]
[image]
[image]

5. Double tap does not work but performing a rest command via JLink brings the device back into DFU mode. (`nrfjprog -f nrf52 --reset`)

charlesportwoodii, 2020-06-29:
That all looks right. It's odd you can't get into the bootloader through with the double tap reset. Reset is wired directly into the nRF chipset, and if the bootloader runs initially it should just "work".

Looking back on my notes here: https://github.com/kaidyth/nrf52_bootloader/issues/7 I used the neopixel lib as a test harness for Arduino on the Feather Express nRF52840. I know that zip in particular won't work cause of the SD version, so I can try to recompile it with the bootloader in debug mode and see what spits out. I'm almost certain that something is just off with the app signing process but won't be able to check until I see what the bootloader complains about in DEBUG mode.

I ordered a Nano 33 BLE and can make a formal definition for it as well so you aren't using the generic one - DEBUG builds take up a lot more space than they ought to.

SaltedBlowfish, 2020-06-29:
@charlesportwoodii Here is the log dump from the JLink RTT Viewer: https://gist.github.com/SaltedBlowfish/fa2aa2e9554099cb4a581053c62b66d1

charlesportwoodii, 2020-06-29:
There's a RAM conflict happening in the application with SD7 - the hung state you're seeing is a hard-fault.

SDK17/SD7 required the RAM regions to be adjusted (see here https://github.com/kaidyth/nrf52_bootloader/commit/92e8e98e79b41377a8051d2d26b19526f3fe45a1#diff-d1bd30e579a6b094610b56c45d32d548R9.)

The Adafruit board lib is still using s140_sd6, whereas the bootloader has RAM reserved for s140_sd7. https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/291 is the outstanding issue on the nRF52 Arduino item to handle upgrading the SoftDevice - it looks like it isn't implemented yet.

A few options for you to try:
1. Adjust the RAM regions for https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/linker/nrf52840_s140_v6.ld to as follows:
    ```
    FLASH (rx) : ORIGIN = 0x27000, LENGTH = 0xc9000 
    RAM (rwx) :  ORIGIN = 0x20002280, LENGTH = 0x3dd80
    uicr_bootloader_start_address (r) : ORIGIN = 0x00000FF8, LENGTH = 0x4
    ```

I can't find the SD7 changelog at the moment, but I pulled these values from an SD7 compatible app I have working on a custom board that I knows works.

Effectively the bootloader is trying to reserve some extra space the application writes into. Increasing the RAM space in the Feather linker file should allocate enough space for the application to run.

2. If you're running Arduino code, downgrade the bootloader to 0.4.0 (https://github.com/kaidyth/nrf52_bootloader/releases/tag/0.4.0) then re-flash the application using 0xB6 as the SoftDevice version

Either option should work, but I'm pretty confident that's the problem. I don't run Arduino code ever on these chips - I would expect that Adafruit upgrade their bootloader regions for others to benefit from this. It's a problem with being an early adopter for SD7 (If you call since September 2019 as early adopting).

Let me know if that works for you. With SD6 on the Feather you should be able to flash the zip file I used as a test harness here: https://github.com/kaidyth/nrf52_bootloader/issues/7 as a quick and dirty test and see the neopixel change colors.

charlesportwoodii, 2020-06-29:
Thinking about ways you can use this with your custom board...

You can fork the repo from 0.4.0, the copy the generic board definition over and run make that way. No guarantees if the docker build still works from that for back, but assuming you can download all the right libs from before Setember it should compile and run you Arduino code sans BLE stuff just fine

SaltedBlowfish, 2020-06-30:
@charlesportwoodii Brilliant! Thank you. I went with option 1. I had a suspicion it was a memory issue but I don't know enough about how to actually find the right regions to modify them.

Your help has been very appreciated. Here's a gif of the lights flashing on my custom board (not a NeoPixel, just RGB LED):

[image]

closed 2020-06-30 by SaltedBlowfish