kaidyth/nrf52_bootloader #21: The nordic cli-tools download in .travis/before_install.sh results in an empty file

jrobeson, 2022-03-30:
I'm not sure how you figured out the post data in a way that is equivalant to "get latest version" which is what I think you were going for.

`wget https://www.nordicsemi.com/api/sitecore/Products/DownloadPlatform --post data=fileid=8F19D314130548209E75EFFADD9348DB -O cli-tools.tar`

results in:
```bash
file cli-tools.tar
cli-tools.tar: empty
```

charlesportwoodii, 2022-03-30:
Thanks for bringing this to my attention @jrobeson. The script grabbed the version at the time it was published - not necessarily the latest. Nordic's changed their website up a bit since I last touched this ...

You should be able to download it from here now: https://www.nordicsemi.com/-/media/Software-and-other-downloads/Desktop-software/nRF-command-line-tools/sw/Versions-10-x-x/10-15-4/nrf-command-line-tools-10.15.4_Linux-amd64.tar.gz

If you need a platform specific variant their website has more direct links here: https://www.nordicsemi.com/Products/Development-tools/nrf-command-line-tools/download

I've updated the build script with a refreshed URL in 29cdd5cefea818c43fb3c12a598b644f38edb126.

jrobeson, 2022-03-30:
did you actually test that? the internal directory structure is totally different.  there's no more `$unpacked_tarball/mergehex/mergehex`, but rather `$unpacked_tarball/bin/mergehex`.

I ended up ignoring the container and doing it locally, since i installed nrfjprog locally anyways.

charlesportwoodii, 2022-03-30:
Sorry I haven't had an opportunity to test anything yet. This links are grabbed directly from Nordic real quick.

PR's are welcome =)

jrobeson, 2022-03-30:
I only mentioned the testing, because as far as i can see, your commit just won't work. 

Ths is my only my second time using docker (actually podman), so I'm not that comfortable with it yet.

charlesportwoodii, 2022-04-03:
I actually publish the docker image, so the one-line command to build your own bootloader would just be. You don't need to actually do anything other than this. The build script is just to build the Docker image locally which isn't strictly necessary.

```
docker run -v${PWD-.}:/app --env BOARD=<BOARD_NAME> DEBUG=1 kaidyth_dfu/toolchain:latest
```

A lot of this scripting was written before Nordic offered debian packages - I'd have to take a look at whats changed in their scripting

charlesportwoodii, 2022-04-03:
All of this has been updated in the provided Docker image and the README updated. Additional the nrf-utils packages has been updated in the Dockerfile.

closed 2022-04-03 by charlesportwoodii